raburton/esp8266 #5: Bug in rboot-ota.c:upgrade_recvcb()

gschmottlach, 2015-06-05:
There appears to be a small bug in the upgrade_recvcb() function of rboot-ota.c. On some servers it appears that the first chunk just details information about the file that will be downloaded (e.g. does not contain any actually rom data). Subsequent chunks actually contain the data. So the check below:
	
if (upgrade->totallength == 0) {

is not a good check because when using a simple Python server to serve up the roms the totallength is still set to zero (0) after the first chunk of the download has been processed (e.g. length -= (ptrData - pusrdata) == 0). So the second chunk isn't processed correctly. My fix was to key off the "sumlength" field that will be zero prior to the first chunk and non-zero on subsequent chunks.

if (upgrade->sumlength == 0 ) {

Perhaps not the most elegant fix but it seems to work for me. I've been able to integrate the rboot OTA mechanism into my Sming-based project.

dranger003, 2015-06-05:
This is fixed by this pull request:
https://github.com/raburton/esp8266/pull/1

raburton, 2015-06-09:
thanks for the feedback, just returned from holiday and found a load of notifications on github (I didn't get any email notifications, I must work out how to fix that), I'll have a look at this as soon as I get a second, but working late tonight...

commit de21714b (2015-06-09): fix failed update with only header in first recv
closes #1 closes #4 closes #5
closed 2015-06-09 by raburton