raburton/esp8266 #12: NodeMCU v2 and SDK v1.2.0: add -fm dio to esptool.py in readme.txt

lrozema, 2015-07-16:
I have rboot working on NodeMCU v2 with SDK v1.2.0. The most important thing for the ESP-12 used on NodeMCU v2 is to add -fm dio to the esptool.py command line. I think this might be worth warning for in readme.txt?

raburton, 2015-07-17:
Can you confirm which device you are referring to. There is no nodemcu v2, do you mean v1.0 (the original being v0.9)?

lrozema, 2015-07-17:
https://www.antratek.nl/nodemcu-v2-lua-based-esp8266-development-kit

lrozema, 2015-07-17:
In English: https://www.antratek.com/nodemcu-v2-lua-based-esp8266-development-kit

raburton, 2015-07-17:
Thanks, that's the v1.0 devkit. Although anyone with these devices should already know they need to use the dio this option (it won't be specific to rboot), I'll add a note to the readme because they are popular dev boards.

commit 6132a2c5 (2015-07-17): mention extra flash options closes #12
closed 2015-07-17 by raburton