raburton/esp8266 #13: Make file error

karincalab, 2015-07-25:
Hi,

Thanks for your great project. But I couldn't get it worked. I always get this error

FW build/rboot-hex2a.h
make[1]: /opt/esp-open-sdk/esp8266/esptool2/esptool2.exe: Command not found

I am not a linux guy, but I am using a ubuntu VM. I know the esptool2.exe is there but it is not compiling it.

Could you please help on this?

Thanks.

daviddpd, 2015-07-25:
In the raburton/esp8266 checkout directory from git ...
```shell
cd esptool2
make
cd ../rboot
```

Then in rboot ... modify the makefile (or probably also at the top-level, but this should give you the gist)

```patch
diff --git a/rboot/Makefile b/rboot/Makefile
index 9d18527..b7d2cb3 100644
--- a/rboot/Makefile
+++ b/rboot/Makefile
@@ -3,9 +3,9 @@
 # https://github.com/raburton/esp8266
 #

-ESPTOOL2 ?= D:/Projects/esp8266/esptool2/Release/esptool2.exe
+ESPTOOL2 ?= ../esptool2/esptool2

-XTENSA_BINDIR ?= C:/xtensa-lx106-elf/bin
+XTENSA_BINDIR ?= /home/vagrant/esp-open-sdk/xtensa-lx106-elf/bin
 CC := $(addprefix $(XTENSA_BINDIR)/,xtensa-lx106-elf-gcc)
 LD := $(addprefix $(XTENSA_BINDIR)/,xtensa-lx106-elf-gcc)
```

raburton, 2015-07-25:
As above, and remember that if you're on linux the compiled esptool2 will not have have an extension (so remove .exe)

karincalab, 2015-07-25:
Thanks for your quick responce, I am new ESP and Ubuntu too. Now I got the this error and after a few try I could not get it work.

Building esptool2 firmware tool
make[1]: Entering directory `/opt/esp-open-sdk/esp8266/esptool2'
CC esptool2.c
gcc -O2 -c esptool2.c -o esptool2.o
CC esptool2_elf.c
gcc -O2 -c esptool2_elf.c -o esptool2_elf.o
LD esptool2
gcc -o esptool2 esptool2.o esptool2_elf.o
make[1]: Leaving directory `/opt/esp-open-sdk/esp8266/esptool2'
Building rBoot boot loader
make[1]: Entering directory `/opt/esp-open-sdk/esp8266/rboot'
CC rboot-stage2a.c
LD build/rboot-stage2a.elf
FW build/rboot-hex2a.h
make[1]: execvp: /opt/esp-open-sdk/esp8266/esptool2: Permission denied
make[1]: *** [build/rboot-hex2a.h] Error 127
make[1]: Leaving directory `/opt/esp-open-sdk/esp8266/rboot'
make: *** [rboot] Error 2

raburton, 2015-07-25:
> execvp: /opt/esp-open-sdk/esp8266/esptool2: Permission denied

You appear to be trying to execute the esptool2 directory instead of the esptool2 compiled within that directory. You probably want to be running /opt/esp-open-sdk/esp8266/esptool2/esptool2

raburton, 2015-08-08:
Closing, no bug and no further feedback provided.

closed 2015-08-08 by raburton