raburton/esp8266 #14: Order of operations - NTP before mDNS breaks mDNS

daviddpd, 2015-08-10:
Strange issue I found using the NTP client as attempting to register on mDNS. 

When NTP is called before mDNS, mDNS fails.  If NTP is called after mDNS, it works just fine.  I suspect that there is an issue with the SDK mDNS or UDP/epconn functions.  

Submitting it here as more as FYI/documentation of the interaction.

http://bbs.espressif.com/viewtopic.php?f=7&t=908&p=3110#p3110

raburton, 2015-08-12:
Interesting, thanks. I will follow the post and see what happens.

raburton, 2015-08-20:
Did the response in the linked forum thread help you fix the problem?

daviddpd, 2015-08-21:
Yeah ... I just posted a reply.  In short, of course it was my code.   Because the way I waited for ntp to complete, ipconfig.ip.addr would be undefined when going to initialize mdns_info *info.    So the info->ipAddr was not set correctly.

No strictly about ifconfig being local, but it being undefined when calling the mDNS setup.

raburton, 2015-08-21:
Super!

closed 2015-08-21 by raburton