raburton/esp8266 #15: rBoot in Sming

crosofg, 2015-08-16:
How do I implement rBoot in any of the examples in Sming framework? Can you tell me the steps involved?

crosofg, 2015-08-16:
I tried to write the rom0.bin and rom1.bin file using esptool.py --port COM7 write_flash -fs 8m 0x00000 rboot.bin 0x02000 rom0.bin 0x82000 rom1.bin 0xfc000 blank4.bin. upon restarting  I am getting this as error.
Fatal exception (0):
epc1=0x40100000, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

raburton, 2015-08-18:
duplicate discussion in [forum] (http://www.esp8266.com/viewtopic.php?f=12&t=3115&start=50#p26204) and here, i have replied to your forum post

closed 2015-08-18 by raburton

raburton, 2015-08-29:
Although this issue is already closed, it's worth mentioning here for anyone who finds it that there is now a sample sming/rboot project in the repository.