raburton/esp8266 #18: Error on compile

crosofg, 2015-08-28:
Hi
I am receiving error upon compiling.
gcc -O2 -Wall -c esptool2.c -o esptool2.o
In file included from esptool2.c:26:
esptool2.h:33:20: stdint.h: No such file or directory
In file included from esptool2_elf.h:27,
                 from esptool2.c:27:
elf.h:25: error: parse error before "Elf32_Addr"
elf.h:25: warning: type defaults to `int' in declaration of `Elf32_Addr'
elf.h:25: warning: data definition has no type or storage class
elf.h:26: error: parse error before "Elf32_Off"
elf.h:26: warning: type defaults to `int' in declaration of `Elf32_Off'
elf.h:26: warning: data definition has no type or storage class
elf.h:27: error: parse error before "Elf32_Half"

This was building sometime back.

raburton, 2015-08-28:
Problem is that you're missing stdint.h, what platform and compiler are you using? The only change that has been made to esptool2 recently was the addition of the -Wall option, so perhaps it was happily ignoring this defaulting before. Easiest fix might be to remove that option.

crosofg, 2015-08-28:
I am using Windows 10 and Mingw. I removed -Wall , still the same error.

raburton, 2015-08-28:
That's odd, as I say the only recent change (since June) was the addition of that one CFLAG. Has anything changed on your system recently (aside from moving to Windows 10, can't imagine that's the issue)? New version of mingw?

At the top of esptool.h you'll find this:
#ifdef WIN32
typedef signed __int8     int8_t;
typedef signed __int16    int16_t;
typedef signed __int32    int32_t;
typedef unsigned __int8   uint8_t;
typedef unsigned __int16  uint16_t;
typedef unsigned __int32  uint32_t;
#else
#include <stdint.h>
#endif

Change the "#ifdef WIN32" to "#if defined WIN32 || defined __MINGW32__" or failing that just make the typedefs unconditional and remove the include of stdint.h.

raburton, 2015-08-28:
fixed the markdown, it was messing up the code to add

raburton, 2015-08-31:
Did this work?

crosofg, 2015-08-31:
No it didnt work in windows env. I installed the virtual box and setup a linux env. The esptool2 compiled and generated .o files.  Then I changed all the paths in the subdirectory's makefile. (From c:/espressif  to /opt/esp-open-sdk) and ran the root make to compile everything.

raburton, 2015-08-31:
I must admit I've now moved my development to Linux. Getting everything to work on windows in the first place took more effort, but I thought it was probably still easier than setting up a Linux dev system in virtual box. When I started to work with Sming it just wasn't playing ball so I decided the investment in time was probably better spent on setting up a Linux image. It didn't actually take much effort at all, I should have done it sooner! Your problem is still odd though, because nothing changed in the code that would explain why it used to work then stopped. I'll close this issue and if anyone else hits it and wants to persist with windows I can look into it further then.

closed 2015-08-31 by raburton