raburton/esp8266 #22: undefined reference to rBoot

crosofg, 2015-09-13:
How do I resolve this? 
c:/tools/sming/Sming/SmingCore/Boards.h:17: undefined reference to `rBootHttpUpdate::rBootHttpUpdate()'

raburton, 2015-09-13:
I'll need more info than that. What are you compiling when you get this error?

crosofg, 2015-09-14:
I am trying to compile the Basic_rboot example. 
11:20:06 **** Build of configuration Sming for project Basic_rBoot ****
make all 
OC out/build/libmain2.a
make -C /C/tools/sming/Sming/rboot
make[1]: Entering directory `/C/tools/sming/Sming/rboot'
CC rboot-stage2a.c
LD /c/tools/Sming/Basic_rBoot/out/build/rboot-stage2a.elf
FW /c/tools/Sming/Basic_rBoot/out/build/rboot-hex2a.h
CC rboot.c
LD /c/tools/Sming/Basic_rBoot/out/build/rboot.elf
FW /c/tools/Sming/Basic_rBoot/out/firmware/rboot.bin
make[1]: Leaving directory `/C/tools/sming/Sming/rboot'
C+ app/application.cpp
C+ /C/tools/sming/Sming/appinit/user_main.cpp
CC /C/tools/sming/Sming/rboot/appcode/rboot-api.c
CC /C/tools/sming/Sming/rboot/appcode/rboot-bigflash.c
AR out/build/app_app.a
LD out/build/app_0.out
out/build/app_app.a(application.o): In function `OtaUpdate_CallBack(bool)':
c:/tools/sming/Sming/SmingCore/Boards.h:17: undefined reference to `rBootHttpUpdate::rBootHttpUpdate()'
c:/tools/sming/Sming/SmingCore/Boards.h:17: undefined reference to `rBootHttpUpdate::addItem(int, String)'
c:/tools/sming/Sming/SmingCore/Boards.h:17: undefined reference to `rBootHttpUpdate::setCallback(void (*)(bool))'
out/build/app_app.a(application.o):(.text._Z9OtaUpdatev+0x28): undefined reference to `rBootHttpUpdate::start()'
out/build/app_app.a(application.o): In function `OtaUpdate()':
C:\tools\Sming\Basic_rBoot/app/application.cpp:30: undefined reference to `rBootHttpUpdate::rBootHttpUpdate()'
C:\tools\Sming\Basic_rBoot/app/application.cpp:39: undefined reference to `rBootHttpUpdate::addItem(int, String)'
C:\tools\Sming\Basic_rBoot/app/application.cpp:48: undefined reference to `rBootHttpUpdate::addItem(int, String)'
C:\tools\Sming\Basic_rBoot/app/application.cpp:61: undefined reference to `rBootHttpUpdate::setCallback(void (*)(bool))'
C:\tools\Sming\Basic_rBoot/app/application.cpp:61: undefined reference to `rBootHttpUpdate::start()'
out/build/app_app.a(application.o): In function `serialCallBack(Stream&, char, unsigned short)':
C:\tools\Sming\Basic_rBoot/app/application.cpp:128: undefined reference to `m_printf'
C:\tools\Sming\Basic_rBoot/app/application.cpp:128: undefined reference to `spiffs_mount_manual'
out/build/app_app.a(application.o): In function `init()':
C:\tools\Sming\Basic_rBoot/app/application.cpp:129: undefined reference to `m_printf'
C:\tools\Sming\Basic_rBoot/app/application.cpp:129: undefined reference to `spiffs_mount_manual'
collect2.exe: error: ld returned 1 exit status
make: *** [out/build/app_0.out] Error 1

raburton, 2015-09-14:
Ok, presumably you're working with the feature/rboot branch of Sming from either anakod or my fork. In which case you need to recompile libsming because rBootHttpUpdate is now part of Sming core, not the sample.

crosofg, 2015-09-14:
How do I do that? Would just "make" from Sming folder do that ?

raburton, 2015-09-14:
Yes, in run make in the Sming/Sming folder (one more Sming down from the folder containing the samples).

crosofg, 2015-09-14:
Upon doing that I am getting another bunch of errors. 
C+ Libraries/TFT_ILI9163C/TFT_ILI9163C.cpp
C+ Libraries/Ultrasonic/Ultrasonic.cpp
C+ Libraries/WS2812/WS2812.cpp
C+ Libraries/si4432/si4432.cpp
In file included from Wiring/../include/user_config.h:27:0,
                 from Wiring/Arduino.h:6,
                 from Libraries/si4432/si4432.h:30,
                 from Libraries/si4432/si4432.cpp:27:
Libraries/si4432/si4432.cpp: In member function 'void Si4432::readAll()':
Libraries/si4432/si4432.cpp:400:98: error: iteration 7u invokes undefined behavior [-Werror=aggressive-loop-optimizations]
     (int ) allValues[i+12], (int ) allValues[i+13], (int ) allValues[i+14], (int ) allValues[i+15]
                                                                                                  ^
system/include/esp_systemapi.h:32:48: note: in definition of macro 'debugf'
 #define debugf(fmt, ...) m_printf(fmt"\r\n", ##__VA_ARGS__)
                                                ^
Libraries/si4432/si4432.cpp:394:21: note: containing loop
  for (byte i = 0; i < 0x7f; i+=16)
                     ^
cc1plus.exe: all warnings being treated as errors
make: *** [out/build/Libraries/si4432//si4432.o] Error 1

raburton, 2015-09-14:
Branch feature/rboot of anakod/Sming compiles cleanly here. I suggest a `git pull` and `git checkout feature/rboot` to make sure you have clean version of the latest source. Failing that I suggest opening an issue against Sming, because this error isn't rBoot related (it's failing in Libraries/si4432/si4432.cpp).

closed 2015-09-14 by crosofg