raburton/esp8266 #24: Second reset  fails

vortex314, 2015-09-21:
I created a .bin and loaded into flash, it contains the rboot image and the executable. The first time I push the reset button the code starts up correctly. When I push the reset button a second time, the bootloader tells it didn't find any correct rom image. I guess the first reset creates a config entry at 0x1000, which the second reset tries to use. 

Project can be found here : https://github.com/vortex314/esp_gtw

I'm using a nodemcu module 32mbit. 

Extracing bin and flashing : 
../tools/esptool.py elf2image ../Debug/$PROJECT
./esptool2 -debug -bin -boot2 -4096 -dio -40 ../Debug/$PROJECT $PROJECT.bin .text .data .rodata
../tools/esptool.py --p /dev/ttyUSB0 -b 921600 write_flash  -ff 40m -fm dio -fs 32m 0x02000 esp_gtw.bin 0x00000 rboot.bin

LD image address
MEMORY
{
  dport0_0_seg :                        org = 0x3FF00000, len = 0x10
  dram0_0_seg :                         org = 0x3FFE8000, len = 0x14000
  iram1_0_seg :                         org = 0x40100000, len = 0x8000
/*  irom0_0_seg :                         org = 0x40240000, len = 0x3C000 */
  irom0_0_seg :                       	org =  0x40202010, len = 0x7C000 
}

closed 2015-09-21 by vortex314

morganrallen, 2015-09-22:
Can you give some detail on what you did to fix this problem? I am seeing a similar issue.

vortex314, 2015-09-22:
I was re-using code from the esp_mqtt project ( https://github.com/tuanpmt/esp_mqtt ) which saves his configuration at an absolute address in flash ( 3C000 ) , which has become with rboot the middle of the code. Commenting out the CFG_Save for the moment , solved it.