raburton/esp8266 #26: FOTA example

fvpalha, 2015-09-24:
Hi Richard.

I am developping  with [Unofficial Development Kit for Espressif ESP8266](https://github.com/CHERTS/esp8266-devkit) and I want to add FOTA feature in our project.

I am studing the [rboot-sampleproject](https://github.com/raburton/esp8266/tree/master/rboot-sampleproject), but this error occurs:
```
mingw32-make.exe -f C:/Trabalho/raburton-esp8266-master/rboot-sampleproject/Makefile all 
CC main.c
In file included from main.c:16:0:
rboot-ota.h:12:23: fatal error: rboot-api.h: No such file or directory
compilation terminated.
mingw32-make.exe: *** [build/main.o] Error 1
```
[image]

Do you have a tip?

Best regards.

fvpalha, 2015-09-24:
Hi Richard.

I do this:
```
3) Symlink or copy rboot.h, rboot-api.h and rboot-api.c in to this directory.
4) Edit the Makefile to set the paths to the SDK and esptool2.
5) Set WIFI_SSID & WIFI_PWD as env vars or in the makefile.
```
and the result was:
```
mingw32-make.exe -f C:/Trabalho/raburton-esp8266-master/rboot-sampleproject/Makefile all 
CC main.c
main.c: In function 'wifi_config_station':
main.c:49:2: error: stray '\' in program
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
  ^
main.c:49:2: error: stray '\' in program
<command-line>:0:13: error: 'dd' undeclared (first use in this function)
main.c:49:31: note: in expansion of macro 'WIFI_SSID'
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
                               ^
<command-line>:0:13: note: each undeclared identifier is reported only once for each function it appears in
main.c:49:31: note: in expansion of macro 'WIFI_SSID'
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
                               ^
main.c:49:2: error: stray '\' in program
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
  ^
main.c:49:2: error: stray '\' in program
<command-line>:0:16: error: 'wrt' undeclared (first use in this function)
main.c:49:31: note: in expansion of macro 'WIFI_SSID'
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
                               ^
main.c:49:2: error: stray '\' in program
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
  ^
main.c:49:2: error: stray '\' in program
main.c:49:2: error: stray '\' in program
main.c:49:2: error: stray '\' in program
main.c:50:2: error: stray '\' in program
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
  ^
main.c:50:2: error: stray '\' in program
<command-line>:0:19: error: stray '@' in program
main.c:50:35: note: in expansion of macro 'WIFI_PWD'
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
                                   ^
main.c:50:2: error: stray '\' in program
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
  ^
main.c:50:2: error: stray '\' in program
<command-line>:0:12: error: 'ESP8266' undeclared (first use in this function)
main.c:50:35: note: in expansion of macro 'WIFI_PWD'
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
                                   ^
main.c:50:2: error: stray '\' in program
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
  ^
main.c:50:2: error: stray '\' in program
<command-line>:0:19: error: stray '@' in program
main.c:50:55: note: in expansion of macro 'WIFI_PWD'
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
                                                       ^
main.c:50:2: error: stray '\' in program
  os_strcpy(&stationConf.password, WIFI_PWD, os_strlen(WIFI_PWD));
  ^
main.c:50:2: error: stray '\' in program
mingw32-make.exe: *** [build/main.o] Error 1
```

raburton, 2015-09-24:
Did you copy the files in to the sample directory, as per the instructions in the readme? Also remmeber to set SDK_BASE. If you do it compiles fine here with UDK:
```
D:\Projects\esp8266\rboot-sampleproject>ls
Makefile              rboot-api.c           rom0.ld
blank4.bin            rboot-api.h           rom1.ld
eagle.rom.addr.v6.ld  rboot-ota.c           uart.c
license.txt           rboot-ota.h           uart.h
main.c                rboot.h               uart_register.h
main.h                readme.txt            user_config.h

D:\Projects\esp8266\rboot-sampleproject>SET SDK_BASE=c:\Espressif\ESP8266_SDK_120

D:\Projects\esp8266\rboot-sampleproject>make
CC main.c
CC rboot-api.c
CC uart.c
CC rboot-ota.c
LD rom0.elf
FW rom0.bin
LD rom1.elf
FW rom1.bin

D:\Projects\esp8266\rboot-sampleproject>
```

raburton, 2015-09-24:
>mingw32-make.exe -f C:/Trabalho/raburton-esp8266-master/rboot-sampleproject/Makefile all 
CC main.c
main.c: In function 'wifi_config_station':
main.c:49:2: error: stray '\' in program
  os_strcpy(&stationConf.ssid, WIFI_SSID, os_strlen(WIFI_SSID));
  ^

Best guess here you have some odd characters in WIFI_SSID & WIFI_PWD.

fvpalha, 2015-09-24:
Hi Richard.

I edited the Makefile:
```
#
# Makefile for rBoot sample project
# https://github.com/raburton/esp8266
#

# use wifi settings from environment or hard code them here
WIFI_SSID ?= "dd-wrt"
WIFI_PWD  ?= "ESP8266@"

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?= c:/Espressif/xtensa-lx106-elf/bin

#SDK_BASE   ?= /opt/esp-open-sdk/sdk
SDK_BASE	?= c:/Espressif/ESP8266_SDK
SDK_LIBDIR  = lib
SDK_INCDIR  = include

ESPTOOL2     ?= ../esptool2/esptool2
FW_SECTS      = .text .data .rodata
FW_USER_ARGS  = -quiet -bin -boot2
```

and SET the environment variable SDK_BASE

```
SDK_BASE	?= c:/Espressif/ESP8266_SDK
```
But this not solve the problem.

raburton, 2015-09-24:
Which problem? You appear to have already got past the original problem you reported.

That whole page of errors you posted next appears to be complaining about the variables WIFI_SSID & WIFI_PWD. I assume you have some none standard characters in them, which probably need to be escaped to become valid C strings. e.g. if your password has a \ in it it'll need to be replaced with \\, etc.

fvpalha, 2015-09-24:
Hi Richard.

I changed the Makefile, I removed the "" in:
```
WIFI_SSID ?= dd-wrt
WIFI_PWD  ?= ESP8266@
```
and now this occurs:
```
mingw32-make.exe -f C:/Trabalho/raburton-esp8266-master/rboot-sampleproject/Makefile all 
CC main.c
CC rboot-api.c
CC rboot-ota.c
CC uart.c
LD rom0.elf
FW rom0.bin
process_begin: CreateProcess(NULL, ../esptool2/esptool2 -quiet -bin -boot2 build/rom0.elf firmware/rom0.bin .text .data .rodata, ...) failed.
make (e=2): O sistema não pode encontrar o arquivo especificado.

mingw32-make.exe: *** [firmware/rom0.bin] Error 2
```

fvpalha, 2015-09-24:
Thank you Richard.

I opened the prompt, and:

```
Setting environment for using Unofficial Development Kit for Espressif ESP8266 t
ools.

C:\Espressif\examples>cd C:\Trabalho\raburton-esp8266-master\esptool2

C:\Trabalho\raburton-esp8266-master\esptool2>make
CC esptool2.c
gcc -O2 -Wall -c esptool2.c -o esptool2.o
esptool2.c: In function 'CreateHeaderFile':
esptool2.c:214: warning: format '%08x' expects type 'unsigned int', but argument
 3 has type 'Elf32_Addr'
esptool2.c:235: warning: format '%08x' expects type 'unsigned int', but argument
 4 has type 'Elf32_Addr'
esptool2.c:235: warning: format '%d' expects type 'int', but argument 6 has type
 'Elf32_Word'
CC esptool2_elf.c
gcc -O2 -Wall -c esptool2_elf.c -o esptool2_elf.o
LD esptool2
gcc -o esptool2 esptool2.o esptool2_elf.o

C:\Trabalho\raburton-esp8266-master\esptool2>cd C:\Trabalho\raburton-esp8266-mas
ter\rboot-sampleproject

C:\Trabalho\raburton-esp8266-master\rboot-sampleproject>make
CC main.c
CC rboot-api.c
CC rboot-ota.c
CC uart.c
LD rom0.elf
FW rom0.bin
LD rom1.elf
FW rom1.bin

C:\Trabalho\raburton-esp8266-master\rboot-sampleproject>
```

Soon I will send new doubts.

Best regards.

closed 2015-09-24 by raburton