raburton/esp8266 #27: OTA - Firmware update failed!

fvpalha, 2015-09-24:
Hi Richard.

I edited the rboot-ota.h:
```
// ota server details
#define OTA_IP { 192, 168, 0, 42 }
#define OTA_PORT 8080
#define OTA_ROM0 "rom0.bin"
#define OTA_ROM1 "rom1.bin"
```

I opened the terminal:

```
ip
ip: 192.168.1.108, mask: 255.255.255.0, gw: 192.168.1.1
ota
Updating...
Firmware update failed!
```

The server (192.168.0.42) find the ESP8266 (192.168.1.108).

```
Microsoft Windows [versão 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Todos os direitos reservados.

C:\Users\palha>ping 192.168.1.108

Disparando 192.168.1.108 com 32 bytes de dados:
Resposta de 192.168.1.108: bytes=32 tempo=29ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=48ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=71ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=93ms TTL=255

Estatísticas do Ping para 192.168.1.108:
    Pacotes: Enviados = 4, Recebidos = 4, Perdidos = 0 (0% de
             perda),
Aproximar um número redondo de vezes em milissegundos:
    Mínimo = 29ms, Máximo = 93ms, Média = 60ms

C:\Users\palha>
```

What is wrong? How I can debug?

raburton, 2015-09-24:
Is your router connecting the two subnets correctly? Can you reach the webserver OK on that port from the other subnet and download the file? Are the files on the root of the webserver? Add some debug to see where it's failing.

fvpalha, 2015-09-25:
Hi Richard.

I changed the server subnet to 192.168.1.*:
```
Microsoft Windows [versão 6.1.7601]
Copyright (c) 2009 Microsoft Corporation. Todos os direitos reservados.

C:\Users\palha>ipconfig

Configuração de IP do Windows

Adaptador de Rede sem Fio Conexão de Rede sem Fio:

   Sufixo DNS específico de conexão. . . . . . :
   Endereço IPv6 de link local . . . . . . . . : fe80::9017:61e0:dd4b:6967%14
   Endereço IPv4. . . . . . . .  . . . . . . . : 192.168.1.107
   Máscara de Sub-rede . . . . . . . . . . . . : 255.255.255.0
   Gateway Padrão. . . . . . . . . . . . . . . : 192.168.1.1
C:\Users\palha>
C:\Users\palha>ping 192.168.1.108

Disparando 192.168.1.108 com 32 bytes de dados:
Resposta de 192.168.1.108: bytes=32 tempo=57ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=80ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=101ms TTL=255
Resposta de 192.168.1.108: bytes=32 tempo=21ms TTL=255

Estatísticas do Ping para 192.168.1.108:
    Pacotes: Enviados = 4, Recebidos = 4, Perdidos = 0 (0% de
             perda),
Aproximar um número redondo de vezes em milissegundos:
    Mínimo = 21ms, Máximo = 101ms, Média = 64ms

C:\Users\palha>
```

I opened the terminal, and:
```
connect
wifi connecting...
network retry, status: 1
ip: 192.168.1.108
ask: 255.255.255.0, gw: 192.168.1.1
ipk: 255.255.255.0, gw: ip
ip: 192.168.1.108, mask: 255.255.255.0, gw: 192.168.1.1
ota
Firmware update failed!
ware update faota
Updating...
Firmware update failed!
```

Now I will add debug points.

fvpalha, 2015-09-25:
Hi Richard.

I changed the operation mode of Apache to online, now it's working.

Congratulations by the project.

raburton, 2015-09-25:
>I changed the operation mode of Apache to online, now it's working.

That will help!

closed 2015-09-25 by raburton