raburton/esp8266 #29: Help edit the Makefile

fvpalha, 2015-09-25:
Hi Richard.

Please, I need a help to build the ROM.

I am trying to add the target "otaall".

```
#############################################################
#
# Root Level Makefile
#
# (c) by CHERTS <sleuthhound@gmail.com>
#
#############################################################

BUILD_BASE	= build
FW_BASE		= firmware

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?= c:/Espressif/xtensa-lx106-elf/bin

# base directory of the ESP8266 SDK package, absolute
SDK_BASE	?= c:/Espressif/ESP8266_SDK

# esptool path and port
SDK_TOOLS	?= c:/Espressif/utils
ESPTOOL		?= $(SDK_TOOLS)/esptool.exe
ESPPORT		?= COM1
ESPBAUD		?= 115200

# BOOT = none
# BOOT = old - boot_v1.1
# BOOT = new - boot_v1.3+
BOOT?=none
# APP = 0 - eagle.flash.bin + eagle.irom0text.bin
# APP = 1 - user1.bin
# APP = 2 - user2.bin
APP?=0
# SPI_SPEED = 20MHz, 26.7MHz, 40MHz, 80MHz
SPI_SPEED?=40
# SPI_MODE: QIO, QOUT, DIO, DOUT
SPI_MODE?=QIO
# SPI_SIZE: 256KB, 512KB, 1024KB, 2048KB, 4096KB
SPI_SIZE?=512
#SPI_SIZE?=1024

# use wifi settings from environment or hard code them here
WIFI_SSID ?= dd-wrt
WIFI_PWD  ?= ESP8266@

ifeq ($(BOOT), new)
    boot = new
else
    ifeq ($(BOOT), old)
        boot = old
    else
        boot = none
    endif
endif

ifeq ($(APP), 1)
    app = 1
else
    ifeq ($(APP), 2)
        app = 2
    else
        app = 0
    endif
endif

[... 300 lines cut ...]
	@$(ESPTOOL2) $(FW_USER_ARGS) $^ $@ $(FW_SECTS)

$(OTA_BUILD_DIR):
	@mkdir -p $@

```

But the result is:

```
16:49:55 **** Build of configuration Default for project TMS_ESP8266_OTA ****
mingw32-make.exe -f C:/Trabalho/TMS_ESP8266/branches/r48_ota/Makefile otaall 
mkdir -p firmware
mkdir -p firmware/upgrade
LD rom0.elf
C:\Espressif\ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x3b0): undefined reference to `user_init'
C:\Espressif\ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_rf_pre_init':
(.irom0.text+0x5f4): undefined reference to `user_init'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [/rom0.elf] Error 1
```

Thank you.
Best regards.

raburton, 2015-09-26:
That Makefile is huge mess, why not just use the one from the sample project. It'll work for most scenarios, and if it needs modifying it'll be a lot easier because it's much simpler.

fvpalha, 2015-09-28:
Hi Richard.

Thank you by tip.

That is the Makefile from the sample project, it works in sample:
```
#
# Makefile for rBoot sample project
# https://github.com/raburton/esp8266
#

# use wifi settings from environment or hard code them here
WIFI_SSID ?= dd-wrt
WIFI_PWD  ?= ESP8266@

# Base directory for the compiler
XTENSA_TOOLS_ROOT ?= c:/Espressif/xtensa-lx106-elf/bin

#SDK_BASE   ?= /opt/esp-open-sdk/sdk
SDK_BASE	?= c:/Espressif/ESP8266_SDK
SDK_LIBDIR  = lib
SDK_INCDIR  = include

ESPTOOL2     ?= ../esptool2/esptool2
FW_SECTS      = .text .data .rodata
FW_USER_ARGS  = -quiet -bin -boot2

#ifndef XTENSA_BINDIR
#CC := xtensa-lx106-elf-gcc
#LD := xtensa-lx106-elf-gcc
#else
#CC := $(addprefix $(XTENSA_BINDIR)/,xtensa-lx106-elf-gcc)
#LD := $(addprefix $(XTENSA_BINDIR)/,xtensa-lx106-elf-gcc)
#endif

# select which tools to use as compiler, librarian and linker
CC	:= $(XTENSA_TOOLS_ROOT)/xtensa-lx106-elf-gcc
AR	:= $(XTENSA_TOOLS_ROOT)/xtensa-lx106-elf-ar
LD	:= $(XTENSA_TOOLS_ROOT)/xtensa-lx106-elf-gcc

BUILD_DIR = build
FIRMW_DIR = firmware

SDK_LIBDIR := $(addprefix $(SDK_BASE)/,$(SDK_LIBDIR))
SDK_INCDIR := $(addprefix -I$(SDK_BASE)/,$(SDK_INCDIR))

LIBS    = c gcc hal phy net80211 lwip wpa main pp
CFLAGS  = -Os -g -O2 -Wpointer-arith -Wundef -Werror -Wno-implicit -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls  -mtext-section-literals  -D__ets__ -DICACHE_FLASH
LDFLAGS = -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static

SRC		:= $(wildcard *.c)
OBJ		:= $(patsubst %.c,$(BUILD_DIR)/%.o,$(SRC))
LIBS		:= $(addprefix -l,$(LIBS))

ifneq ($(WIFI_SSID), "")
	CFLAGS += -DWIFI_SSID=\"$(WIFI_SSID)\"
endif
ifneq ($(WIFI_PWD), "")
	CFLAGS += -DWIFI_PWD=\"$(WIFI_PWD)\"
endif

.SECONDARY:
.PHONY: all clean

C_FILES = $(wildcard *.c)
O_FILES = $(patsubst %.c,$(BUILD_DIR)/%.o,$(C_FILES))

all: $(BUILD_DIR) $(FIRMW_DIR) $(FIRMW_DIR)/rom0.bin $(FIRMW_DIR)/rom1.bin

$(BUILD_DIR)/%.o: %.c %.h
	@echo "CC $<"
	@$(CC) -I. $(SDK_INCDIR) $(CFLAGS) -o $@ -c $<

$(BUILD_DIR)/%.elf: $(O_FILES)
	@echo "LD $(notdir $@)"
	@$(LD) -L$(SDK_LIBDIR) -T$(notdir $(basename $@)).ld $(LDFLAGS) -Wl,--start-group $(LIBS) $^ -Wl,--end-group -o $@

$(FIRMW_DIR)/%.bin: $(BUILD_DIR)/%.elf
	@echo "FW $(notdir $@)"
	@$(ESPTOOL2) $(FW_USER_ARGS) $^ $@ $(FW_SECTS)

$(BUILD_DIR):
	@mkdir -p $@

$(FIRMW_DIR):
	@mkdir -p $@

# esptool path and port
SDK_TOOLS	?= c:/Espressif/utils
ESPTOOL		?= $(SDK_TOOLS)/esptool.exe
ESPPORT		?= COM1
ESPBAUD		?= 115200
flash: all
	$(ESPTOOL) -p $(ESPPORT) -b $(ESPBAUD) write_flash -fs 8m 0x00000 $(FIRMW_DIR)/rboot.bin 0x02000 $(FIRMW_DIR)/rom0.bin 0x82000 $(FIRMW_DIR)/rom1.bin 0xfc000 $(FIRMW_DIR)/blank4.bin

clean:
	@echo "RM $(BUILD_DIR) $(FIRMW_DIR)"
	@rm -rf $(BUILD_DIR)
	@rm -rf $(FIRMW_DIR)
```

And the result is:

```
mingw32-make.exe -f C:/Trabalho/TMS_ESP8266/branches/r48_ota/Makefile all 
LD rom0.elf
C:\Espressif\ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_uart_wait_tx_fifo_empty':
(.irom0.text+0x3b0): undefined reference to `user_init'
C:\Espressif\ESP8266_SDK/lib\libmain.a(app_main.o): In function `user_rf_pre_init':
(.irom0.text+0x5f4): undefined reference to `user_init'
collect2.exe: error: ld returned 1 exit status
mingw32-make.exe: *** [build/rom0.elf] Error 1
```

raburton, 2015-09-28:
Looks like your code doesn't have a user_init function in it.

fvpalha, 2015-10-02:
Hi Richard.

Congratulations by the great project.

I suggested to @cherts to add the rboot-sampleproject into [Unofficial Development Kit for Espressif ESP8266](https://github.com/CHERTS/esp8266-devkit).

To make easy add the FOTA resource in a new project based on others devkit examples, for newbies like me, it's necessary edit the Makefile.

[Please, help me change the files/folders and the Makefile](https://github.com/CHERTS/esp8266-devkit/issues/21)

Best regards.

closed 2015-11-21 by raburton