raburton/esp8266 #30: restart on rBoot-sampleproject does not always restart

whyameye, 2015-11-03:
I've been playing with rBoot-sampleproject and overall it works great. OTA update is smooth. Sometimes, however, restart is not successful. Here is an example output from the usb serial port with the original code, unaltered in function from the git repository:
```help
    V1 available commands
  help - display this message
  ip - show current ip address
  connect - connect to wifi
  restart - restart the esp8266
  switch - switch to the other rom and reboot
  ota - perform ota update, switch rom and reboot
  info - show esp8266 info

ip
  network status: 255
connect
       wifi connecting...
network retry, status: 1
ip: 192.168.43.5
restart
       Restarting...

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
at this point the 8266 hangs indefinitely until I reset it again.

ESP12-e on nodeMCU dev 1.0 board. latest commit 978b3cf  (13 days ago at the time of this post).

raburton, 2015-11-03:
Did you set flash to dio mode? I think the esp12e requires this.

whyameye, 2015-11-04:
If that were the problem, wouldn't it never work? Software reset works 95% of the time. When it doesn't work, a hard reset always works. So the code is fine in the flash memory.

raburton, 2015-11-04:
From your log it doesn't look like it's getting as far as running rBoot, not much happens before the first message is (or should be) output, so I don't think that's the problem. The link to Pete's blog is an issue which I've discussed with him previously. After a serial flash the next reboot never works. But after a hard reset it should work normally again after that and you should be able to reset to your hearts content.

closed 2015-11-21 by raburton

zhivko, 2016-09-09:
Maybe this is reason:
https://github.com/esp8266/Arduino/tree/master/doc/faq#i-have-observed-a-case-when-esprestart-doesnt-work-what-is-the-reason-for-that