raburton/esp8266 #33: give rBoot it's own repo

morganrallen, 2015-12-03:
I've been using rBoot for a couple months now and I'm pretty much sold on it.

Now I'm just getting to the point I'm cleaning up my first project using it and with any
luck will be using it in an upcoming client project. With that it would be super handy
to have rBoot at a git submodule, but that requires it being in its own repo.

Hopefully this isn't an issue, thanks for you work on this, it's vastly improved my workflow.

spants, 2015-12-04:
I'm also looking to tidy up my IoTBox project Richard - learning about using git submodules - so this would be handy for me too! Thanks for your help..

closed 2015-12-05 by raburton

morganrallen, 2015-12-05:
Fantastic! Thanks @raburton

commit 9ccc8229 (2015-12-05): separate esptool2, rboot and rboot-sample in own repos, closes: #33